kriswiner/LSM9DS1 #3: temperature conversion

juniskane, 2015-01-27:
The line:

temperature = ((float) tempCount/256. + 25.0); // Gyro chip temperature in degrees Centigrade

only gives you (incorrect) temperatures from narrow range +17 to +33 ⁰C (because chip temp output registers are 12-bits only: -2048...+2047)

Where did you get the 256 value? For ThingSee device I'm using a different divisor (when converting to integer in hundredths of ⁰C):

	/* 12-bit value, max. range -2048..+2047. Sensitivity according to datasheet
	 * is quoted at 16 LSB/⁰C (Typ.), so we use that. 
	 * Zero raw value = 25 ⁰C.
	 */
         enum { LSM9DS1_TEMPERATURE_PRECISION=100 };

	*int_temper = *raw_temper * LSM9DS1_TEMPERATURE_PRECISION / 16 +
		25 * LSM9DS1_TEMPERATURE_PRECISION;

For some other ST parts, the datasheets don't even bother to quote the sensitivity or even the zero point!

kriswiner, 2015-01-27:
This must be a remnant from the LSM9DS0 or some other sensor. The sensitivity should be 16/LSB as in the data sheet.

I didn't pay much attention to this since I have a 24-bit pressure/temperature sensor on the board.

huirad, 2016-02-05:
I agree with sensitivity 16/LSB based on my own measurements (comparison with temperature output of pressure sensor on same board and with ambient temperature at startup) .
At least for the LSM9DS1 on my SenseHat module, the offset is rather 27.5 than 25.0

I use the following in my code

#define LSM9DS1_TEMP_SCALE   16.0
#define LSM9DS1_TEMP_BIAS    27.5
static float conv_temp(int16_t raw_temp)
{
    return raw_temp / LSM9DS1_TEMP_SCALE + LSM9DS1_TEMP_BIAS;
}

kriswiner, 2016-02-05:
The nominal temperature sensor response is supposed to be 25 degrees Celsius
when the counts are zero, but there is some variation and ST's guidance is
to calibrate with external standard for accuracy as you have done. The
intent of the temperature sensor is to keep track of the (gyro) die
temperature and compensate if necessary. It is not intended as an
environmental sensor.

<https://github.com/kriswiner/LSM9DS1/issues/3#issuecomment-180548286> .
<https://github.com/notifications/beacon/AGY1qua-7WDS6VNTQGs5VsEGo-WH47y-ks5
phQD6gaJpZM4DXuVE.gif>