kriswiner/LSM9DS1 #8: Not completely working for me

ibycneemoh, 2017-08-16:
Hi Kris, I have been trying to make my LSM9DS1 with your code. I just have a few questions.

I am using madwick filter on an arduino uno.
`MadgwickQuaternionUpdate(ax, ay, az, gx * PI / 180.0f, gy * PI / 180.0f, gz * PI / 180.0f,  -mx,  my, mz);`

`if (readByte(LSM9DS1M_ADDRESS, LSM9DS1M_STATUS_REG_M) & 0x08) {  // check if new mag data is ready  
    readMagData(magCount);  // Read the x/y/z adc values
    
    // Calculate the magnetometer values in milliGauss
    // Include factory calibration per data sheet and user environmental corrections
    mx = (float)magCount[0]*mRes; // - magBias[0];  // get actual magnetometer value, this depends on scale being set
    my = (float)magCount[1]*mRes; // - magBias[1];  
    mz = (float)magCount[2]*mRes; // - magBias[2];   
  }`

In the above code why do you comment our the bias on magnetometre?

Secondly with my yaw has a lot of problems. I place the board on a breadboard and leave it on a place surface. i then rotate it by the z axis. but i can not see any changes. 

below pic shows me moving in an eight shape and then putting the sensor to rest. 
[image]

the format is ax,ay,az,gx,gy,gz,mx,my,mz, roll,pitch, yaw. The sensor is flat on a breadboard on a tale and i am rotating it about the z axis.

Is there something i am doing wrong or some registers not outputting something? is there a way to factory reset the sensor?

finally i am using this sketch to visualise my imu.
[https://github.com/PaulStoffregen/NXPMotionSense/blob/master/OrientationVisualiser/OrientationVisualiser.pde](url)

it just doesnt seem to be doing anything.I rotate it then put it to rest and it reacts so slow. when i move it, it gets confused, goes the wrong way then back to the the other way.

Hope you can help fix this. 

Kind Regards

kriswiner, 2017-08-16:
> OrientationVisualiser/OrientationVisualiser.pde <http://url>

Not sure what is wrong but you are not getting correct mag values so
everything else is moot.

Are you using Sparkfun's code or what?

ibycneemoh, 2017-08-16:
Hi Kris,
i have also used the wire library and made changes to user fast mode.

Thanks for your quick response. I am using a sparkfun breakout board but i have made tiny changes to your code. I have commented out most of the MS5611 as the board doesn't have it.
If you can help that will be great.
The code is below. Why do you comment out the magnetometer bias?

The formatting isn't great below but i hope it makes sense where i made tiny changes.

`/** LSM9DS1_MS5611_t3 Basic Example Code
  by: Kris Winer
  date: November 1, 2014
  license: Beerware - Use this code however you'd like. If you
  find it useful you can buy me a beer some time.

  Demonstrate basic LSM9DS1 functionality including parameterizing the register addresses, initializing the sensor,
  getting properly scaled accelerometer, gyroscope, and magnetometer data out. Added display functions to
  allow display to on breadboard monitor. Addition of 9 DoF sensor fusion using open source Madgwick and
  Mahony filter algorithms. Sketch runs on the 3.3 V 8 MHz Pro Mini and the Teensy 3.1.

  This sketch is intended specifically for the LSM9DS1+MS5611 Add-on shield for the Teensy 3.1.
  It uses SDA/SCL on pins 17/16, respectively, and it uses the Teensy 3.1-specific Wire library i2c_t3.h.
  The MS5611 is a simple but high resolution pressure sensor, which can be used in its high resolution
  mode but with power consumption od 20 microAmp, or in a lower resolution mode with power consumption of
  only 1 microAmp. The choice will depend on the application.

  SDA and SCL should have external pull-up resistors (to 3.3V).
  4K7 resistors are on the LSM9DS1+MS5611 Teensy 3.1 add-on shield/breakout board.

  Hardware setup:
  LSM9DS1Breakout --------- Arduino
  VDD ---------------------- 3.3V
  VDDI --------------------- 3.3V
  SDA ----------------------- A4
  SCL ----------------------- A5
  GND ---------------------- GND

  Note: The LSM9DS1 is an I2C sensor and can use the Arduino Wire library.
  Because the sensor is not 5V tolerant, we are using either a 3.3 V 8 MHz Pro Mini or a 3.3 V Teensy 3.1.
  We have disabled the internal pull-ups used by the Wire library in the Wire.h/twi.c utility file.
  We are also using the 400 kHz fast I2C mode by setting the TWI_FREQ  to 400000L /twi.h utility file.
*/
#include "Wire.h"
//include <i2c_t3.h>
#include <SPI.h>
//#include <Adafruit_GFX.h>
//#include <Adafruit_PCD8544.h>

// Using NOKIA 5110 monochrome 84 x 48 pixel display
// pin 7 - Serial clock out (SCLK)
// pin 6 - Serial data out (DIN)
// pin 5 - Data/Command select (D/C)
// pin 3 - LCD chip select (SCE)
// pin 4 - LCD reset (RST)
//Adafruit_PCD8544 display = Adafruit_PCD8544(7, 6, 5, 3, 4);

// See MS5611-02BA03 Low Voltage Barometric Pressure Sensor Data Sheet
#define MS5611_RESET      0x1E
#define MS5611_CONVERT_D1 0x40
#define MS5611_CONVERT_D2 0x50
#define MS5611_ADC_READ   0x00

// See also LSM9DS1 Register Map and Descriptions, http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00103319.pdf
//
// Accelerometer and Gyroscope registers
#define LSM9DS1XG_ACT_THS	    0x04
#define LSM9DS1XG_ACT_DUR	    0x05
#define LSM9DS1XG_INT_GEN_CFG_XL    0x06
#define LSM9DS1XG_INT_GEN_THS_X_XL  0x07
#define LSM9DS1XG_INT_GEN_THS_Y_XL  0x08
[... 1313 lines cut ...]
  Wire.endTransmission();        // Send the Tx buffer, but send a restart to keep connection alive
  //	Wire.endTransmission(false);       // Send the Tx buffer, but send a restart to keep connection alive
  uint8_t i = 0;
  //        Wire.requestFrom(address, count);  // Read bytes from slave register address
  Wire.requestFrom(address, (size_t) count);  // Read bytes from slave register address
  while (Wire.available()) {
    dest[i++] = Wire.read();
  }         // Put read results in the Rx buffer
}

void printAHRS()
{

  Serial.print("acc(g): "); Serial.print("\t"); Serial.print(ax); Serial.print("\t"); Serial.print(ay); Serial.print("\t"); Serial.print(az); Serial.print("\t");
  Serial.print("gyro(d/s): "); Serial.print("\t"); Serial.print(gx); Serial.print("\t"); Serial.print(gy); Serial.print("\t"); Serial.print(gz); Serial.print("\t");
  Serial.print("mag(G): "); Serial.print("\t"); Serial.print(mx); Serial.print("\t"); Serial.print(my); Serial.print("\t"); Serial.print(mz); Serial.print("\t");

  Serial.print("Orientation: "); Serial.print("\t");
  Serial.print(yaw); Serial.print("\t");  Serial.print(pitch);  Serial.print("\t"); Serial.print(roll); Serial.print("\t");

  Serial.print("Quat: "); Serial.print("\t");
  Serial.print(q[0]); Serial.print("\t");  Serial.print(q[1]);  Serial.print("\t"); Serial.print(q[2]); Serial.print("\t"); Serial.println(q[3]);
  //Serial.print(q0); Serial.print("\t");  Serial.print(q1);  Serial.print("\t"); Serial.print(q2); Serial.print("\t"); Serial.println(q3);
}
`

kriswiner, 2017-08-16:
> technical/document/datasheet/DM00103319.pdf
> quaternions_and_Euler_angles which has additional links.

Not sure what to make of this, I can;t diagnose your code for you. You are
outputting accel in gs and mag in Gauss. hard to tell what is going on.

ibycneemoh, 2017-08-16:
is there a particular reason for working in mg and mG? I thought the Madwick needed the values in standard units?

Any idea how i can reset this to factory settings? please

kriswiner, 2017-08-16:
The madwick filter uses normalized units for accel and mag (but no gyro),
so it doesn't matter, but if you want a human to diagnose your problem
better to use units that humans can understand.

Reset to factory settings??? Ask Sparkfun...

ibycneemoh, 2017-08-16:
Thank you Kris, i will try to run your exact code and post the results with sensor at rest after calibration. Thank you for your help.

ibycneemoh, 2017-08-17:
Hi Kris,

I have run the your code and just formatted the order of printing. it is printing in mg and mG. 
I perform the calibration and put the sensor at rest.

Also my magnetometer calibration results, the bias can all be negative and can be a mix of positive and negative results. not sure if this has any impact at all. 

I am also scratching my head about doing a self test for the magnetometer. using CTRL_REG1_M (20h). There is no information on how to go about this and what to expect, any ideas?

Kind Regards

[image]

ibycneemoh, 2017-08-17:
Just to add print format for about results is 
ax, ay, az, gx, gy, gz, mx, my, mz, yaw, pitch, roll, q[0], q[1], q[2], q[3]

kriswiner, 2017-08-17:
OK, that looks much better.

ibycneemoh, 2017-08-23:
Hello Kris,

I have updated from Arduino Uno to a Teensy 3.6. I am thinking i am doing the right calibration as it is your code i am using. 

below are my calibration results. Am i doing something wrong? The yaw is just not very responsive and seems off.

In your on of your responses you mentioned my magnetometer readings are wrong, could you please elaborate on this.?

Also in your code, you do not apply any bias to the reading. is it beacuse you qwrite the biases to the sensors?

Here when you read the magbias is commented out.
`if (readByte(LSM9DS1M_ADDRESS, LSM9DS1M_STATUS_REG_M) & 0x08) {  // check if new mag data is ready
    readMagData(magCount);  // Read the x/y/z adc values

    // Calculate the magnetometer values in milliGauss
    // Include factory calibration per data sheet and user environmental corrections
    mx = (float)magCount[0] * mRes; // - magBias[0];  // get actual magnetometer value, this depends on scale being set
    my = (float)magCount[1] * mRes; // - magBias[1];
    mz = (float)magCount[2] * mRes; // - magBias[2];`

This one is after calibration and placing the sensor down
[image]

this is with the sensor at rest after calibratin.
[image]

kriswiner, 2017-08-23:
How did you verify that the mag is calibrated?

ibycneemoh, 2017-08-23:
I haven't used any external user interface. My apologies for my ignorance and assumptions. Do you have any suggestions?

Another questions is you do not do the soft iron calibrations like you do for the 9250. any particular reason why?

kriswiner, 2017-08-23:
You need to do something like this:

https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration

Might not be in my sketch since it was posted years ago before I started
doing mag calibration in the sketches.

ibycneemoh, 2017-09-01:
Hi Kris, while i was looking for a tool to do verify my calibration i found this tool. I have used this tool to calibrate and i got these values (completely different from what i what getting from the code).  https://www.pjrc.com/store/prop_shield.html

[image]

My issue is now to follow this example https://github.com/adafruit/Adafruit_AHRS/blob/master/examples/ahrs_fusion_usb/ahrs_fusion_usb.ino

In your code you apply offset to registers OFFSET_X_REG, OFFSET_Y_REG and OFFSET_Z_REG.
Can i write the values of this registers to Zero and apply the offsets via software as done in the adafruit example? Can this eliminate the whole magnetometre calibration in your code?

kriswiner, 2017-09-01:
Yes and yes.

ibycneemoh, 2017-09-01:
Thanks you very much will let you know how it goes

ibycneemoh, 2017-09-03:
Hi Kris,
So after testing the calibration and replacing the tools i have still not managed to get any sensible results. My yaw is very unstable, slow to respond and not rotating well still. a 90 degree turn  for instant can not be produced. I have tried to put the results from the calibration and tried to do the calibration at start as well. Results are not that different.

after twitching with the figures in your code i am getting a rate between 711Hz to 2040Hz. Here is how i am setting some parametres from your code (rather the only lines i have played with changing stuff back and forth).
 
Please help if you can spot any errors.

If possible i can email the whole code with tiny changes i made but it is mainly based on yours. I am using the Teensy 3.6.

any wise words of advice please?

`float beta = sqrt(3.0f / 4.0f) * GyroMeasError;   // compute beta`

`uint8_t OSR = ADC_4096;      // set pressure amd temperature oversample rate
uint8_t Gscale = GFS_245DPS; // gyro full scale
uint8_t Godr = GODR_238Hz;   // gyro data sample rate
uint8_t Gbw = GBW_med;       // gyro data bandwidth
uint8_t Ascale = AFS_2G;     // accel full scale
uint8_t Aodr = AODR_238Hz;   // accel data sample rate
uint8_t Abw = ABW_50Hz;      // accel data bandwidth
uint8_t Mscale = MFS_4G;     // mag full scale
uint8_t Modr = MODR_20Hz;    // mag data sample rate
uint8_t Mmode = MMode_HighPerformance;  // magnetometer operation mode
float aRes, gRes, mRes;      // scale resolutions per LSB for the sensors`

`deltat = ((Now - lastUpdate) / (1000000.0f)); // set integration time by time elapsed since last filter update`

`  if (delt_t > 0.25) { // update LCD once per half-second independent of read rate//25ns
`
I have tried both filters.

How can i make the filter react quicker and settle quicker? I have tried to play with beta sticking random values from 0.041 to 0.6. 

[image]

kriswiner, 2017-09-03:
With the Teensy 3.6 you should be getting fusion rates of ~50 kHz.
Something is wrong...

ibycneemoh, 2017-09-03:
can i email you my piece of code so you can try it your end? please.

kriswiner, 2017-09-03:
No thanks

ibycneemoh, 2017-09-03:
Ok, thank you for the help i will keep on trying.

kriswiner, 2017-09-03:
What happens if you just run one of my sketches without fiddling with it?

ibycneemoh, 2017-09-03:
Thank you for all the help, i appreciate

This is what i get, please see screenshots below.

I have taken the filter code and stuck it right at the bottom. Only addition i have made. I only get 3.6 max from running this sketch. I am using the sparkfun board and it only contains the lsm9ds1 only. I have 3 of these and have swapped them multiple times

[image]

[image]

ibycneemoh, 2017-09-03:
I am using this code here 
https://github.com/kriswiner/LSM9DS1

kriswiner, 2017-09-03:
OK, this looks pretty good to me.

Did you calibrate the sensors?

I mean I see that the accel data is not 0, 0, 1 as I would expect for lying
flat on a table motionless.

If you turn the sensor first face up and then face down, do you get Mz up =
-Mz down?

The fusion rate is rather low, but maybe you are not using the interrupts?
On the other hand, it is fast enough you should be getting pretty accurate
and stable heading.

The only way I know not to is:

1) not properly calibrated
2) not feeding sensor data into Madgwick algorithm properly.

If

kriswiner, 2017-09-03:
That code has no magnetometer calibration...

ibycneemoh, 2017-09-03:
I am slightly confused when you say that code has no calibration. At start up it asks you to move the sensor in 8 position which i understand is hard iron calibration. I can not see any soft iron correction though.

do you have another version of this code for this particular device that uses interrupts and the calibration you expect?

No i had a basic I2C setup. No interrupts.

I can see an int intPin = 15; line which i am not sure where this pin connects to.

When i turn my sensor Mz up does not equal - Mz Down. Do i need to calibrate first to achieve this or this should be expected behaviour without calibration?

kriswiner, 2017-09-03:
I just noticed that the offset bias correction seems to be commented out
for the mag data in that sketch.

I suspect your mag is not calibrated, and without mag calibration your
heading will be garbage.

This is why I asked if you had calibrated your sensors and how did you know.

ibycneemoh, 2017-09-03:
From lines 897 to 902 in the calibration function for the magnetometre you write the bias to the offset registers. My assumption was this is how you calibrate the magnetometre? Should i uncomment the bias even when you write to the offset registers?

kriswiner, 2017-09-03:
You should check the calibration of your sensors.

If you place the sensor flat and motionless you should see:

0 mg, 0 mg, 1000 mg for xyz accel

and

0 dps, 0 dps, 0 dps for xyz gyro.

Do you?

On the mag, plot Mx vs. My, Mx vs. Mz, and My vs Mx while moving the sensor
all around.

You should see circular clouds of points centered on the origin.

Do you>

Until you can answer yes to these questions your sensors are not calibrated.

ibycneemoh, 2017-09-03:
I see that for the acc and the gyro. I will give another go at the mag

ibycneemoh, 2017-09-03:
In one of my posts above i showed  calibration using the tool, perhaps it is important to note that results are similar?

ibycneemoh, 2017-09-04:
Hi Kris,

So i have reset my offset registers and Mz up = -Mz down. I am applying the offsets in the software rather than writing to the offset registers. 

My calibration has been verified

The problem i have now is with the filter part. it goes to the right angle then quickly goes back up to 360. Any idea what may be happening?