kriswiner/LSM9DS1 #9: Magnetometer Calibration

dmamalis, 2017-12-20:
@kriswiner i have been using your code to make an ecompass and I have had a few problems which I am trying to pin down. Seeing issues on other repos of yours I decided to put some more effort on calibrating the magnetometer before posting.

According to my results, unless I am doing something terribly wrong, the magcal algorithm seems to be inadequate.

First I run it multiple times and found out that the magBias results are completely different. Then I plotted some experiments to make this a bit more concise. In the next screenshots you can see three views of the same experiment.

[image]
[image]
[image]
 
It looks like there is need for both soft and hard iron calibration.

kriswiner, 2017-12-20:
Yes, and the algorithm described here
<https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration>
does a pretty good job of both, you might have to collect data long enough
to get a good sampling of the response surface. You are moving the sensor
to sample the entire 3D sphere during the calibration, aren't you?

If you just leave the sensor flat on a table during the calibration, you
aren't doing it right...

dmamalis, 2017-12-20:
That was a quick reply! Thank you.

I also used the MotionCalc calibration tool to visualize data and I realized that the previous screenshots MUST be wrong, at least with regards to the soft iron issue. I am guessing my gnuplot script is faulty.

Here is the MotionCalc screenshot with shows much better soft iron results
[image]

No I am definitely not leaving it flat but I guess a few more samples might be a better solution.

Last but not least, could you please explain to me why the magBias is commented out in these lines?
https://github.com/kriswiner/LSM9DS1/blob/master/LSM9DS1_MS5611_BasicAHRS_t3.ino#L437

kriswiner, 2017-12-20:
> MS5611_BasicAHRS_t3.ino#L437

Because this sketch is using the mag offset bias registers internal to the
LSM9DS1:

//write biases to accelerometermagnetometer offset registers as counts);
writeByte(LSM9DS1M_ADDRESS, LSM9DS1M_OFFSET_X_REG_L_M, (int16_t) mag_bias[0]
& 0xFF);
writeByte(LSM9DS1M_ADDRESS, LSM9DS1M_OFFSET_X_REG_H_M, ((int16_t)mag_bias[0]
>> 8) & 0xFF);
writeByte(LSM9DS1M_ADDRESS, LSM9DS1M_OFFSET_Y_REG_L_M, (int16_t) mag_bias[1]
& 0xFF);
writeByte(LSM9DS1M_ADDRESS, LSM9DS1M_OFFSET_Y_REG_H_M, ((int16_t)mag_bias[1]
>> 8) & 0xFF);
writeByte(LSM9DS1M_ADDRESS, LSM9DS1M_OFFSET_Z_REG_L_M, (int16_t) mag_bias[2]
& 0xFF);
writeByte(LSM9DS1M_ADDRESS, LSM9DS1M_OFFSET_Z_REG_H_M, ((int16_t)mag_bias[2]
>> 8) & 0xFF);

dmamalis, 2017-12-22:
I changed the samples to 512 to give some extra data and the calibration is much better now. I consider this closed. Thank you.

closed 2017-12-22 by dmamalis