kriswiner/LSM9DS1 #11: raw and pitch change simultaneously

aa18514, 2018-04-27:
Hi,

I can confirm the following: 

a) I have calibrated my magnetometer, accelerometer and gyroscope correctly. 

b) The data being passed in the Mahony and the Madgewick filter is in the correct order. 
(ax, ay, az, gx, gy, gz, -1 * mx, my, mz)

I seem to be getting correct orientation for raw, but as pitch changes from +PI/2 to -PI/2, I also observe a change in yaw by PI radians, as pitch starts approach PI/2 radians, yaw starts getting distorted. I would be grateful if you could please provide some insight. 

Many thanks,

aa18514, 2018-04-27:
I saw in your code, you have accelgyrocalLSM9DS1() and magcalLSM9DS1() functions, I run it and move the chip in figure 8 and take average of the bias values, is it a correct way to do calibration ? will the calibration affect the orientation ?

The real question is:
I believe that the roll, pitch and yaw should be independent to each other. When you rotate the chip in one direction, for example pitch, the other two should stay nearly the same. When I do one complete rotation, yaw and pitch change by the same amount, (for example one complete rotation for yaw by 2 * pi radians also gives the same change for pitch).

kriswiner, 2018-04-27:
Are you sure the max x and y axes are aligned with the accel/gyro x and y
axes? This is from my sketch:

 MadgwickQuaternionUpdate(ax, ay, az, gx*PI/180.0f, gy*PI/180.0f,
gz*PI/180.0f,  -mx,  my, mz);

aa18514, 2018-04-27:
I can also confirm the fusion rate I get on ESP8266 board fluctuates around 3.7 kHz

kriswiner, 2018-04-27:
If you have properly calibrated the sensors then this is the best you can
do.

aa18514, 2018-04-28:
I use motionCal to calibrate my sensors please find my results as follows:                                                                         
[image]

aa18514, 2018-04-28:
to be more specific as pitch goes from 0 degrees to -60 degrees, yaw goes from 0 to -40 degrees.

kriswiner, 2018-04-28:
Something is wrong with either your sensor data (not calibrated) or your
Madgwick function. The sketch I reposited at Github does not do this.

aa18514, 2018-04-29:
Dear Kris,    

Many thanks for your support,   I can confirm the following:

a) The Mahony and the Madgewick functions remain unchanged. 

I believe the problem could potentially come from the calibration. What is interesting is that I tried using the same approach (i.e motionCal to calibrate) with Adafruit NXP 9 DOF (https://learn.adafruit.com/nxp-precision-9dof-breakout?view=all) and that seems to perform fine. 

I believe that motionCal uses microTesla base units, which is why while calibrating lsm9ds1, I tried catering for this and made the appropriate changes, the following lines change from: 

mx = (float)magCount[0] * mRes - magBias[0]
my = (float)magCount[1] * mRes - magBias[1]
mz = (float)magCount[2] * mRes - magBias[2]

to the following: 
mx = mRes * (0.1f * (float) magCount[0] - magBias[0]);
my = mRes * (0.1f *(float) magCount[1] - magBias[1]);
mz = mRes * (0.1f * (float) magCount[2] - magBias[2]);

perhaps I am missing something else?

Best Wishes

kriswiner, 2018-04-29:
> nxp-precision-9dof-breakout?view=all) and that seems to perform fine.

The accel and mag units are normalized anyway in the fusion algorithm so
this is likely not the issue.

This sounds like an NED issue, but your function call looks OK, not sure.
It seems to work for me. Are you sure you are usin the LSM6DS1 and not the
LSM6DS0?

kriswiner, 2018-04-29:
One oother thing, the gyro units have to be converted to radians/sec before
being passed to the fusion algorithm. Are you doing this?

kriswiner, 2018-04-30:
> Dear Kris, I can confirm that I am using the LSM6DS1 using MCU esp8266, I
> can also confirm that the data is being passed as radians/s to the fusion
> algorithm,

Not sure what else I can say. You appear to be doing everything right so it
should work for you as it does for me.

aa18514, 2018-05-18:
Hi Kris,

Many thanks for your reply, I used a new LSM9DS1 chip and followed the same steps and everything seems to work. I consider the thread closed.

closed 2018-05-18 by aa18514