kriswiner/LSM9DS1 #12: Calibration question

oliverFj, 2018-11-19:
Hi.
I am having trouble getting consistent readings from the lsm9ds1, and I think the issue is with my calibrations. 

The thing is, I have had a couple of attempts where the readings have been spot on, but other times there is a lot of drift. So I know it works. 
I have found that the best calibrations are made when I turn the devise around the yaw, pitch and roll, and then back again. You instruct to wave the device in the figure eight, could you perhaps expand on this technique? 

Is there a way I can store the calibrations, or is it nessesary to calibrate it for every use?

kriswiner, 2018-11-19:
The idea is to sample all of 3-space with the movement so the magnetometer
can get readings from every possible direction for the calibration.

If you think of the device with a big arrow going from front to back and
with the device inside a large rubber ball, the point of the arrow has to
be pointed at every location within the surface of the ball during the data
collection phase of the calibration

Not sure how to make it any more clear.

You can always store the calibration data in the Arduino sketch or program
code.

saedelman, 2019-02-28:
@kriswiner - thanks for the sample LSM9DS1 code. I'm using it and trying to determine linear acceleration by removing the gravity vector from the accelerometer data, and this works but only when the sensor is oriented a specific way (e.g. when flat on the table, I'm getting 0.0 as expected) but when oriented other ways, I'm getting 2.0. This tells me that the quaternion equation for vx,vy,vz does not have the correct polarity. Any suggestions?

```
if (readByte(LSM9DS1XG_ADDRESS, LSM9DS1XG_STATUS_REG) & 0x01)
    {  // check if new accel data is ready
    readAccelData(accelCount);  // Read the x/y/z adc values

    // Now we'll calculate the accleration value into actual g's
    ax = (float)accelCount[0]*aRes; // - accelBias[0];  // biases subtracted in transforma()
    ay = (float)accelCount[1]*aRes; // - accelBias[1];
    az = (float)accelCount[2]*aRes; // - accelBias[2];

    // sphere-ize the accel data
    transforma(&ax,&ay,&az);

    // get gravity vector estimate from quaternion
    float vx = 2 * (q[1] * q[3] - q[0] * q[2]);
    float vy = 2 * (q[0] * q[1] + q[2] * q[3]);
    float vz = q[0] * q[0] - q[1] * q[1] - q[2] * q[2] + q[3] * q[3];

    // get magnitude of linear acceleration
    float pk = sqrt(((ax-vx)*(ax-vx)) + ((ay-vy)*(ay-vy)) + ((az-vz)*(az-vz)));
}
```

kriswiner, 2019-02-28:
Have you tried these
<https://github.com/gregtomasch/EM7180_SENtral_Calibration> sketches?

saedelman, 2019-02-28:
@kriswiner - figured it out. I previously swapped the x and y accel, gyro and mag variables in the Mahony fusion algorithm to eliminate sitting near the 180/-180 point when my device is oriented in the proper way in the vehicle (could probably rotate the LSM9DS1 on the PCB). I changed it as follows:

`MahonyQuaternionUpdate(ay, ax, az, gy*M_PI/180.0f, gx*M_PI/180.0f, gz*M_PI/180.0f, my, -mx, mz);`

This required that I change the gravity vector estimate to:

```
float vy = 2 * (q[1] * q[3] - q[0] * q[2]);
float vx = 2 * (q[0] * q[1] + q[2] * q[3]);
float vz = q[0] * q[0] - q[1] * q[1] - q[2] * q[2] + q[3] * q[3];

```
After that, when I rotate the device (slowly) around all axis, the linear acceleration stays around 0.0-0.1 until I wack it against the table. Nice...

kriswiner, 2019-02-28:
Glad you got it to work...