kriswiner/LSM9DS1 #13: Magnetometer data coherence

Anticiptik, 2019-06-27:
Hi, 

I am using a LSM9DS1 on a custom PCB, with your code (thanks a lot btw), for a dead reckoning algorithm. For now I am outputing calculated quaternions for a Python script (https://github.com/thecountoftuscany/PyTeapot-Quaternion-Euler-cube-rotation), that calculates Euler angles and allow a visualization of the board position in 3D. Everything seems to be working fine when I try to manually move the board, yaw, pitch and roll angles values seem correct.

But when I try to put my device in a car and I take a 90° or 180° turn (with a few meters radius), mag values doesn't seem to change: on the visualizer, I can see the board trying to turn, because of the gyroscope input, but it is dampened by magnetometer data that doesn't change, and so it will then slowly go back to magnetometer calculated heading, that is still the same from before the turn... I also tried to calculate the heading by simply using magnetometer data (atan2(my,mx)), and have the same result. Plus, if I use a version of Madgwick algo without mag data, only accel and gyro, the turn is detected. So I know this error is coming from the magnetometer. 

I'm sorry I know this issue might not be directly correlated to your code, but I just wanted to ask if you knew about such an issue and its origin.

Thank you in advance for any help !

kriswiner, 2019-06-27:
This sounds like a combination calibration and hard iron problem.

You can look up the magnetic field in your location using, for example, the
USGS site. In my location, the "z" component of the Earth's magnetic field
is ~430 milliGuass. So when I have adequately calibrated the magnetometer,
I read 430 +/- 10 milliGuass when the sensor is facing up and -430 +/- 10
milliGuass when facing down and this doesn't change if I rotate about the
"z" axis on a flat surface by, say, 90 degrees, etc. In other words, there
are easy ways to test whether the magnetometer is behaving as it should in
an absolute sense.

Now the problem you are having is either there is an extra source of
magnetic field in your lab (too close to the laptop) or in your car (big
hunk of iron). One (or both) of these is causing a big hard iron offset to
your mag values such that the puny ~0.5 Guass magnetic field of Earth is
not registering.

Anticiptik, 2019-06-28:
Thank you for your answer. 

I indeed should have 455 mGauss on my magnetometer z axis (it is the z axis right, when the component is "flat" and facing up ?), as this is the vertical component of the magnetic field at my position, but I have ~ -510 mGauss when facing up and ~ 70 mGauss when facing down, at z axis. And this after having calibrated the magnetometer following your code and waving as it should be during the calibration process... (600 samples). Those values seems very wrong though, so either I'm not doing/mesuring something correctly, either I really have a local mag issue.

These values also seem to change when rotating the component around the z axis, so this would come from soft iron calibration ?