kriswiner/LSM9DS1 #14: Arduino Nano 33 BLE with LSM9DS1 sensor -> problem with implementation Madgwick libary

mataldomen, 2020-02-16:
Hey Kris,

I am writing to You becouse I have big problem with my Arduino board and I can't resolve this from week.

I want to implement Madgwick filter and quaternions into my Arduino Nano 33 BLE board. After that i receive a lot of really crazy numbers without any sense. 

I configured My board with ArduinoLSM9DS1 libary (https://www.arduino.cc/en/Reference/ArduinoLSM9DS1) it's really simply part of code in comparison into yours.  For me is perfect beacouse I am beginner.

In my first use of this board I used data from sensors and I calculated Eulers angles to receive pitch,roll and yaw. I received preety good numbers.

Nextly I want to implement madgwick from here(https://x-io.co.uk/open-source-imu-and-ahrs-algorithms/)

Maybe You know with Your experience why I still cant use Madgwick on my board ? Maybe You done someting before on this board ?

kriswiner, 2020-02-16:
https://github.com/kriswiner/LSM9DS1/blob/master/LSM9DS1_MS5611_BasicAHRS_t3.ino

MichaelFBA, 2020-02-23:
Hi Kris

How did you go translating the example for the nano 33?
I am also struggling a little getting accurate results with the new boards.

Did you want to share some code maybe we can figure it out together?

kriswiner, 2020-02-23:
I did not translate the sketch for the Nano33, I have never used the Nano33.

Did you calibrate the sensors? How?

MichaelFBA, 2020-02-23:
Ah sorry I meant @mataldomen :)

I've managed to get some calibration data from motioncal but reapplying the offsets has been a little difficult. I was hoping to have some realtime calibration methods when the device starts up.

I'll try your methods to see if I can implement it.
Cheers

MichaelFBA, 2020-03-03:
@mataldomen 

You can try this version of the Madgwick lib for the nano 33 ble.
I still need to figure out the calibration step, but might help getting you somewhere if you dont need calibrated data.

```
#include <ArduinoBLE.h>
#include <Arduino_LSM9DS1.h>
#include <MadgwickAHRS.h>

// Madgwick
Madgwick filter;
// sensor's sample rate is fixed at 119 Hz:
const float sensorRate = 119;

float roll, pitch, heading;

// BLE Service
BLEService imuService("917649A0-D98E-11E5-9EEC-0002A5D5C51B"); // Custom UUID

// BLE Characteristic
BLECharacteristic imuCharacteristic("917649A1-D98E-11E5-9EEC-0002A5D5C51B", BLERead | BLENotify, 12);

long previousMillis = 0;  // last timechecked, in ms
unsigned long micros_per_reading, micros_previous;

void setup() {
  Serial.begin(115200);    // initialize serial communication

  pinMode(LED_BUILTIN, OUTPUT); // initialize the built-in LED pin to indicate when a central is connected

  // begin initialization
  if (!IMU.begin()) {
    Serial.println("Failed to initialize IMU!");
    while (1);
  }

  if (!BLE.begin()) {
    Serial.println("starting BLE failed!");
    while (1);
  }

  // Setup bluetooth
  BLE.setLocalName("ArduinoIMU");
  BLE.setAdvertisedService(imuService);
  imuService.addCharacteristic(imuCharacteristic);
  BLE.addService(imuService);

  // start advertising
  BLE.advertise();
  Serial.println("Bluetooth device active, waiting for connections...");

  // start the filter to run at the sample rate:
  filter.begin(119);

  delay(10000);

  Serial.print("Accelerometer sample rate = ");
  Serial.print(IMU.accelerationSampleRate());
  Serial.println(" Hz");
  Serial.println();
  Serial.println("Acceleration in G's");
  Serial.println("X\tY\tZ");
  Serial.print("Gyroscope sample rate = ");
  Serial.print(IMU.gyroscopeSampleRate());
  Serial.println(" Hz");
  Serial.println();
  Serial.println("Gyroscope in degrees/second");
  Serial.println("X\tY\tZ");
  Serial.print("Magnetic field sample rate = ");
  Serial.print(IMU.magneticFieldSampleRate());
  Serial.println(" uT");
  Serial.println();
  Serial.println("Magnetic Field in uT");
  Serial.println("X\tY\tZ");

  micros_per_reading = 1000000 / 119;
  micros_previous = micros();
}

// send IMU data
void sendSensorData() {

  float ax, ay, az; // Acceleration
  float gx, gy, gz; // Gyroscope
  float mx, my, mz; // Magnometer

  // read orientation x, y and z eulers
  IMU.readAcceleration(ax, ay, az);
  IMU.readGyroscope(gx, gy, gz);
  IMU.readMagneticField(mx, my, mz);

  filter.update(gx, gy, gz, ax, ay, az, -mx, my, mz); //for all 3
  roll = filter.getRoll();
  pitch = filter.getPitch();
  heading = filter.getYaw();
  Serial.print("Orientation: ");
  Serial.print(heading);
  Serial.print(" ");
  Serial.print(pitch);
  Serial.print(" ");
  Serial.println(roll);

  // Send 3x eulers over bluetooth as 1x byte array
  float data[3];
  data[0] = heading;
  data[1] = pitch;
  data[2] = roll;
  imuCharacteristic.setValue((byte *) &data, 12);

}

void loop() {
  // wait for a BLE central
  BLEDevice central = BLE.central();

  // if a BLE central is connected to the peripheral:
  if (central) {
    Serial.print("Connected to central: ");
    // print the central's BT address:
    Serial.println(central.address());
    // turn on the LED to indicate the connection:
    digitalWrite(LED_BUILTIN, HIGH);

    // while the central is connected:
    while (central.connected()) {
      unsigned long micros_now;
      micros_now = micros();

      if (micros_now - micros_previous >= micros_per_reading) {
        if (IMU.accelerationAvailable() && IMU.gyroscopeAvailable() && IMU.magneticFieldAvailable()) { // XX
          sendSensorData();
          micros_previous = micros_previous + micros_per_reading;
        }
      }
    }
    // when the central disconnects, turn off the LED:
    digitalWrite(LED_BUILTIN, LOW);
    Serial.print("Disconnected from central: ");
    Serial.println(central.address());
  }
}
```

mataldomen, 2020-03-07:
Hi @MichaelFBA 
Thank You for your help and code. 
Code looks and work really similar which my. I was sure about that numbers which We receive  from Arduino are faulty beacuse I  bad use  Madgwick libary. 
I didnt think before about that sensors without calibration in real time can create so big mistake. Probably You are right.

Do You have any ideas how calibrate them properly ?

MichaelFBA, 2020-03-07:
This repo has examples of how to calibrate the sensor at runtime, we would need to modify it to conform with the Arduino imu API. 

I had a quick look at trying to migrate the code but haven't managed to finish it yet. There is a lot going on.

elsatch, 2020-04-20:
As I was researching this topic, I discover this topic on the Arduino forum that could provide additional insights: https://forum.arduino.cc/index.php?topic=663160.msg4467543#msg4467543

Accoding to the illustriation presented, magnetometer axis are flippped from the ones on accelerometerand gyroscopes. It also seems that rotation is left handed!

kriswiner, 2020-04-20:
Yes, this is all in the data sheet which everone who is using this sensor
should read completely, perhaps any times. The specific orientation of the
sensor axes has to be taken into account in the Madgwick or Mahony fusion
call.

elsatch, 2020-04-22:
Thanks @kriswiner for your response!

Jeremiah1327, 2020-05-09:
Hello @MichaelFBA and @mataldomen , were either of you able to calibrate this sensor for the Arduino Nano 33 BLE? I have been stuck on this for a few days now. I tried changing the sensor axis and still cannot get it to work.

mataldomen, 2020-05-10:
Hello,
Unfortunately I quit my project without solved this problem. When I read about my wrong setup of axis two weeks ago i tried do this properly again and nothing improve.

I cant help You. If You have to set orientation based on quaternions maybe You shoud buy another sesnsor wchich calculating quaternions whithout madgwick libary. On market is big choice but these sesnors are more expensive then our :D

MrStashley, 2020-06-01:
https://www.youtube.com/watch?v=T9jXoG0QYIA

@mataldomen @MichaelFBA  @Jeremiah1327  @elsatch I hope I'm not too late for you guys but I've recently have been working on the same problem and I've found some resources. As you can see from this video it's possible to calibrate the magnetometer without any libraries in a sort of brute force manner by taking a bunch of measurements in different places over a period of time, finding the median x y and z values and shifting them to 0 where median - 0 would be the offset that you could then add to every magnetometer reading and achieve calibration. I haven't looked just yet but I bet there are some 3D plotting libraries to help with this as well. It also seemed implied by the video that accelerometer values and gyroscope values were okay and didn't need to be calibrated, but I'm not sure if that is the case. I have noticed that the accelerometer and gyroscope read a little bit of noise when at rest and I think that disregarding values below some noise threshold might be enough. 

I've also looked into motion cal https://learn.adafruit.com/adafruit-sensorlab-magnetometer-calibration/magnetic-calibration-with-motioncal but I'm struggling to get the sketches to work on the Nano 33 BLE Sense, if some of you guys want to try them, let me know if you can get them to work.

I've also experimented with using this algorithm: https://x-io.co.uk/open-source-imu-and-ahrs-algorithms/ in my receiver program to calculate a quaternion of rotation given the raw sensor data over bluetooth from my arduino, but the end result seems to be very inacurate and some of the axes seem to be wrong.  I haven't calibrated yet so that might be part of my issue, also I have been randomly experimenting with the beta and sample period parameters since I'm not really sure what they're supposed to be, so that also might be a contributing factor lol

but that's where I've gotten so far, let me know if anyone has more information, thanks!

kriswiner, 2020-06-01:
Basic accel/gyro, and mag calibration as well as quaternions are
implemented here
<https://github.com/kriswiner/LSM9DS1/blob/master/LSM9DS1_MS5611_BasicAHRS_t3.ino>.
Just need to make some small changes for your platforms. Not sure why this
is such a mystery....

kriswiner, 2020-06-01:
> Apologies, I’m a bit new to this. I saw that but I was a bit deterred
> because there was a lot of code that I didn’t understand. Is this a library
> that you can get from the arduino store or at least import into the arduino
> editor? And what would we need to change to make it compatible with the
> arduino nano 33 ble sense?

Probably just need to change the Wire methods since i2c_t3.h is Teensy only.

But you can certainly copy the methods for accel/gyro calibration, mag
calibration, and quaternion determination. This sketch is not meant to be a
black box that you can plug into your Nano but rather an illustration of
methods that can be used.

Jeremiah1327, 2020-06-01:
@MrStashley I spent many hours a day for over a week trying to get it to work. I watched a lot of YouTube videos about IMU calibration methods and tried to modify the numerous LSM9DS1 codes. I tried to modify @kriswiner 's code but still could not get it figured out. I bought the Arduino Nano BlE Sense for the purpose of having a built in IMU that would be easy to use. I decided to just try out different IMU/Microcontrollers with working libraries. I simply do not have the time and coding experience to do get it working with the BLE Sense and am not willing to sacrifice more time to get it to work, as I have deadlines for my project. If you end up figuring it out though, that would be awesome.

MrStashley, 2020-06-01:
@Jeremiah1327 ah okay. I'm still doing research and I haven't yet gotten around to trying to get @kriswiner 's code to work, but I have looked at and I see some stuff that seems promising and getting another IMU is a last resort for me because I need the ble chip, but it might come to that for me as well. Which IMU did you end up buying that worked for you?

MrStashley, 2020-06-02:
@kriswiner  do you know how I could implement the writeByte method in a way that would work on my nano 33 ble sense?

kriswiner, 2020-06-02:
This works on the ESP8252 and is pretty generic:

// I2C read/write functions

void writeByte(uint8_t address, uint8_t subAddress, uint8_t data)
{
Wire.beginTransmission(address); // Initialize the Tx buffer
Wire.write(subAddress); // Put slave register address in Tx buffer
Wire.write(data); // Put data in Tx buffer
Wire.endTransmission(); // Send the Tx buffer
}

uint8_t readByte(uint8_t address, uint8_t subAddress)
{
uint8_t data; // `data` will store the register data
Wire.beginTransmission(address); // Initialize the Tx buffer
Wire.write(subAddress); // Put slave register address in Tx buffer
Wire.endTransmission(false); // Send the Tx buffer, but send a restart to
keep connection alive
Wire.requestFrom(address, 1); // Read one byte from slave register address
data = Wire.read(); // Fill Rx buffer with result
return data; // Return data read from slave register
}

void readBytes(uint8_t address, uint8_t subAddress, uint8_t count, uint8_t
* dest)
{
Wire.beginTransmission(address); // Initialize the Tx buffer
Wire.write(subAddress); // Put slave register address in Tx buffer
Wire.endTransmission(false); // Send the Tx buffer, but send a restart to
keep connection alive
uint8_t i = 0;
Wire.requestFrom(address, count); // Read bytes from slave register address
while (Wire.available()) {
dest[i++] = Wire.read(); } // Put read results in the Rx buffer
}

   - © 2020 GitHub, Inc.
   - Terms <https://github.com/site/terms>
   - Privacy <https://github.com/site/privacy>
   - Security <https://github.com/security>
   - Status <https://githubstatus.com/>
   - Help <https://help.github.com>

<https://github.com>

MrStashley, 2020-06-02:
Ah okay. I thought Wire was a library that I might have to replace but my editor seems to think it's fine. The error I'm getting now is that I2C_NOSTOP was not declared in this scope and it seems that it wasn't declared in your code either. Is this a constant that I can define or something native to a certain kind of board?

kriswiner, 2020-06-02:
This is for i2c_t3.h only, replace with false as in the code snippets I
sent you.

MrStashley, 2020-06-02:
ahh okay my bad I didn't notice. Thanks a ton :)

MrStashley, 2020-06-02:
So the writeByte method causes the Mbed OS to crash. It could be an issue with the actual registers used in the calls but the first writeByte call in accelgyrocal and in magcal crashes the OS. I'm sure this problem is way out of my expertise league, and it might just be a compatibility issue that's impossible to fix, but can you think of anything that might be related to the issue that I could look into?

kriswiner, 2020-06-02:
Run! Run away as fast as you can from MBED....

MrStashley, 2020-06-02:
Ah okay that's what I'm finding out lol. It sucks because the Nano 33 ble sense has everything I need for my project in terms of sensors and ble capability. Do you know of any other boards that have similar capabilities but don't have the annoying issue of being compatible with nothing?

kriswiner, 2020-06-02:
Is there not an Arduino core for the Nano33?

https://www.arduino.cc/en/Guide/NANO33BLESense

MrStashley, 2020-06-02:
There is, it's just that none of the calibration libraries are compatible with the board so I haven't been able to calibrate the sensors and I can't get the AHRS algorithm to spit out numbers that makes even close to sense. I've heard that the sensors on the nano 33 ble sense have coordinate planes in wierd directions and that the accelerometer and gyroscope have left handed coordinate planes and I think that might be contributing to my error as well, but I'm not really sure how to fix it.

kriswiner, 2020-06-02:
The Madgwick fusion algorithm required the sensors data be fed in in NED
convention.

Besides, if using the Arduino IDE, you can just run my sketch (with slight
modification) and get all of this.

MrStashley, 2020-06-02:
ah okay thank you for that. I'll look into which axes I need to switch to get NED.

well except that the writeByte function causes the Mbed OS to crash. Do you think that it's an issue with the location of the registers on my board or an issue with the Wire functions?

kriswiner, 2020-06-02:
The Wire functions are written in C++ and are suitable for use with the
Arduino IDE, not C-based MBed.

MrStashley, 2020-06-02:
Well I'm using the arduino IDE to upload programs, but then running them on my ble sense which is running Mbed OS. I guess that's the problem. That seems like such an oversight on the part of the Mbed developers lol, like who uses C-? 
Is there an alternative to Wire, or is my best option to get another board?

kriswiner, 2020-06-02:
I don't know anything about your platform so can't really say. But the link
I sent seems to indicate that there is an LSM6DS1 library for the Aduino
IDE for this Nano33 so I am wondering why you are having anything to do
with MBed at all.

Anyway, the trouble you are having is due to the Nano33 and its programming
environment and not the LSM6DS1, so I don;t expect anything I tell you will
help at this pint.

MrStashley, 2020-06-02:
after making all the accel and gyro axes negative as well as the mz axis in an attempt to get NED convention, I suddenly get vastly different numbers. They're still inaccurate, but they're much closer and I still can play with the gyro scale. Calibration is still going to be a problem but this is promising. Thanks for your help! :)

MrStashley, 2020-06-02:
There is an lsm9ds1 library but it doesn't deal with calibration or anything, it's pretty basic and all it does is read values. I might be able to rig up a calibration in unity tho if I'm lucky. Thanks for your help and I hope I didn't take up too much of your time.

kriswiner, 2020-06-02:
aN, aE, aD, gN, gE, gD, mN, mE, mD is the order the madgwick filter expects
the sensor data. Once you select which board edge is to be North, you
should be able to determine the correspondence between the actual axes
values and NED rather than using trial and error.

kriswiner, 2020-06-02:
There is an LSM6DS1 sketch in my repository that has calibration methods in
it. Not sure why you are not using this...

MrStashley, 2020-06-02:
So I have this diagram: 
[image]
that I believe is from the lsm9ds1 data sheet and I'm not sure how to figure out which side is north on the actual board, but I decided to align the accelerometer and gyroscope with the magnetometer because someone on a forum said that the way the magnetometer was aligned was better, so I used all negative accel and gyro values and -z of the magnetometer to get down. Does that seem reasonable? It made my ahrs values a lot better

MrStashley, 2020-06-02:
okay, thanks. I'll check it out

kriswiner, 2020-06-02:
" I decided to align the accelerometer and gyroscope with the magnetometer"

Is the accel x-axis ligne with the mag x-axis????

No wonder you are getting nonsensical results.

Why not just try NED?

MrStashley, 2020-06-02:
Well how do I get NED? I apologize, I'm very new to this and I thought that I was getting NED by aligning all of the axes and pointing z down. But I see now that the accel and gyro x is aligned with mag y and accel and gyro x is aligned with mag y. Could I account for this by putting gy and ay in place of ax and ay in the update call and vice versa?

MrStashley, 2020-06-02:
accel and gyro y is aligned with mag x**

kriswiner, 2020-06-02:
1) Pick a board edge to point North

2) which accel axis is aligned with this board edge?

3) This is the axis that goes into the first Madgwick sensor data slot, aN.

4) repeat for East and Down, to fill second and third data slots.

5) repeat for gyro and mag

MrStashley, 2020-06-02:
So what I'm struggling with is how to figure out where the lsm9ds1 is on the board and where the axes align

MigSanc, 2020-06-09:
Hi all,

I hope my comment helps you @MrStashley @mataldomen 

I have been playing around today with the Arduino nano 33 BLE sense and I followed @kriswiner instructions, which are standard when using a new IMU. The following image shows the coordinate system that I used which I  confirm is according to the datasheet. You can also check it yourselves by reading the accelerometer, gyroscope, and magnetometer individually.
Example: at rest you should read az = 1 and ax=ay=0, Bz = -uT, and so on for other orientations.

[image]

Thus, you should assume that:
aN = ax, aE = ay, aD = -az
gN = gx, aE = gy, aD = -gz
mN = -mx, aE =my, aD = -mz

After calibrating the magnetometer for the soft and hard iron distortions and applying the Madgwick filter using beta = 0.033 (optimal value according to Madgwick internal report) I was obtaining the Euler angles as they should.

Try it out and see if it works for you.

Best regards to all

MichaelFBA, 2020-06-09:
Can you provide a repo with an example @MigSanc?

MrStashley, 2020-06-09:
@MigSanc ooh nice work and thanks for that awesome diagram. I was able to get my euler angles to be approximately accurate using similar methods but I am still suffering from drift and the algorithm seems to take a very long time to converge unless I use a beta of greater than .1. How were you able to calibrate the magnetometer? Is there a library that you used? 
Thanks and great work!

mnajib76, 2020-07-13:
> This repo has examples of how to calibrate the sensor at runtime, we would need to modify it to conform with the Arduino imu API.
> 
> I had a quick look at trying to migrate the code but haven't managed to finish it yet. There is a lot going on.

hi @MichaelFBA , did you manage migrate the code to use with Nano Ble Sense?

MigSanc, 2020-07-13:
Hi @mnajib76 

There is a good code that you can use:
https://github.com/FemmeVerbeek/Arduino_LSM9DS1

I used some parts of the library above and some and then after aligning the axis to the  NED convention, calibrating and using madgwick filter to calculate quaternions and obtain euler angles, everything worked fined and without drifts.
Give it a try

mnajib76, 2020-07-13:
Hi @MigSanc,

Thanks for your reply. I will try to work on it.. Thanks!

mnajib76, 2020-07-13:
Hi @MigSanc ,

If you don't mind could you explain the step you do to get this done in detail? How did you "aligning the axis to the NED convention"? sorry if this question is too basic to ask but I really new with this and I need a guide to help me do this project.. If you don't mind could you give example how you do this? thanks !

MigSanc, 2020-07-13:
No problem @mnajib76 

So, if you check the LSM9DS1 datasheet (https://www.st.com/resource/en/datasheet/lsm9ds1.pdf), you will see that the axis for each sensor is as follows, correct?
[image]

So, the NED convention states that the axis should be aligned with North (N), East(E), Down(D).
So, choose which face of the arduino BLE you want to be facing north and to which axis does that correspond. From there align all of them to be equal and follow the same convention. As an example:
[image]
In this case, I chose the tip of the arduino board opposed to the USB port connection to be aligned it North. In this case, the x axis from the accelerometer and the gyroscope are already aligned with the convention and the y axis of the magnetometer is as well

aN = ax, gN = gx
aE = ay, gE = gy

But all the others, are not. Example: the z axis of accelerometer, gyroscope and magnetometer should all be pointing "Down", so the new transformation becomes the opposite sign
 aD = -az, gD = -gz, mD = -mZ. 

Then, apply the same principle to all others remaining. Post your transformations here and I will check it later on.

Best regards

mnajib76, 2020-07-18:
Hi @MigSanc !

Thanks for your explanation. Sorry for late respond since I've been busy with other stuff. Anyway, my progress as of now I know how to calibrate the acc,gyr, and mag sensor by using [this](https://github.com/FemmeVerbeek/Arduino_LSM9DS1) (ref 1) guide.. And after calibrate the data , I feed the calibrated data using "Output of Read method" in the "ref 1" documentation into MadgwickAHRS data fusion algorithm which I get from [here](https://github.com/PaulStoffregen/MadgwickAHRS/blob/master/examples/Visualize101/Visualize101.ino) (ref 2) . And I did NED convention as you suggest. Kindly see my code below

`#include <Arduino_LSM9DS1.h>
#include <MadgwickAHRS.h>

Madgwick filter;
unsigned long microsPerReading, microsPrevious;

void setup() {
  
  
  // put your setup code here, to run once:
  Serial.begin(9600);

  filter.begin(238);

  if (!IMU.begin())
  {
    Serial.println("Failed to initialize IMU!");
    exit(1);
  }

   IMU.setAccelFS(3);
   IMU.setAccelODR(4);
   IMU.setAccelOffset(-0.006524, -0.012115, -0.017701);
   IMU.setAccelSlope (1.004801, 0.999822, 0.997208);
   
 
   IMU.setGyroFS(3);
   IMU.setGyroODR(4);
   IMU.setGyroOffset (2.120483, -0.655273, 0.308716);
   IMU.setGyroSlope (1.165733, 1.136184, 1.163230);

   IMU.setMagnetODR(8);  
   IMU.setMagnetOffset(-12.634888, 38.231812, -2.759399);
   IMU.setMagnetSlope (1.594882, 1.571591, 1.637965);

    
  microsPerReading = 1000000 / 238;
  microsPrevious = micros();

}

void loop() {
  // put your main code here, to run repeatedly:
  float x_acc, y_acc, z_acc;
  float x_gyr, y_gyr, z_gyr;
  float mx, my, mz;
  float roll, pitch, heading;
  unsigned long microsNow;

  microsNow =  micros();
  if(microsNow-microsPrevious>=microsPerReading)
  {
      if (IMU.accelerationAvailable()&&IMU.gyroscopeAvailable()&& IMU.magneticFieldAvailable()){
        IMU.readAcceleration(x_acc, y_acc, z_acc);
        IMU.readGyroscope(x_gyr, y_gyr, z_gyr);
        IMU.readMagneticField(mx, my, mz);
      }

      filter.update(x_gyr,y_gyr,x_acc,y_acc,-z_acc,-z_gyr,-mx,my,-mz);
      
      roll = filter.getRoll();
      pitch = filter.getPitch();
      heading = filter.getYaw();
      
      Serial.print(heading);
      Serial.print(",");
      Serial.print(pitch);
      Serial.print(",");
      Serial.println(roll);
      
      microsPrevious = microsPrevious + microsPerReading;
  } 
  
}
`

I put 238Hz as the Output data rate from documentation. However I still not get the reading correctly. In image below is when the sensor stay on flat surface...Hmmm...where did I missed?

[image]

mnajib76, 2020-07-18:
Opss.. I got this error in my code 

So the correct one is ``filter.update(x_gyr,y_gyr,-z_gyr,x_acc,y_acc,-z_acc,-mx,my,-mz);`. But still now work :(`

kriswiner, 2020-07-18:
gyro data needs to be in radians for the filter.

mnajib76, 2020-07-18:
Hi @kriswiner , thanks for highlighting that. However from the code example seems like it suggest convert to degree/s

[image]

It seems my code show improvement on its values which is make sense after re calibrate. But another issue is I found the value of pitch and roll are fluctuate when I rotate and the reading slowly settle down after the sensor stay level.

[image]

And the roll and pitch data seem fluctuate/overshoot as below

[image]

Now I'm stuck.lol..Do you have any Idea what happens @kriswiner ?

Thanks and regards

MigSanc, 2020-07-18:
Hi @mnajib76 

The code that you have posted before in ( https://github.com/kriswiner/LSM9DS1/issues/14#issuecomment-660429239 ) reads gyroscope data in degrees per second and the Madgwick library that you posted already converts from degrees/s to radians/s, so that is taken care. 
The fluctuations might be due to the beta value or/and the time updating readings to the Madgwick filter. 
So, the beta value acts as a proportional gain to compensate for the error on the measurements. If the beta value is too high, the angles will jitter a lot. If beta value is too low then it will fail to compensate and will take time to reach a stable value. 
Concerning the timing, remember that you are setting the filter to 238Hz right? But, are you truly updating the values on the filter at that rate?
In my opinion, I would make some tests, and divide the problem into minor ones. Personally, it helps me to checkpoints and find out where is the error. These are the checkpoints that I do when using IMU's:

1) after calibration, are the accelerometer values correct to the position? if flat --> az = 1 and ax=ay=0; If tuned 90º--> az=0 and ax=1 or 0 and ay=0 or 1 depending the axis that you rotated
2) after calibration, if the sensor is still, the gyroscope should be around 0. You are not moving, so it should not report any movement.
3) go through the magnetometer data. After calibration plot a cartesian chart with 3 sets of data (like below)
[image]
If your magnetometer data is centered around 0 and all circles have the same radius, then your magnetometer calibration was successful
4) Time to move to the frequency analysis: Print the looping frequency that you read and send data to the madgwick filter. 
If you are setting the filter.begin() to expected 238samples per second you need to achieve this rate. I suggest that you modify the Madgwick library to have as an argument the time that you are looping through measurements, for two reasons:
 - the sample frequency will be dynamic instead of static
 - consequently, the quaternion filter will be more accurate that way
5) start playing with beta value and understand how beta plays a role in compensating errors in measurements.
6) confirm that the euler angles are as expected. Some versions of the madgwick library don't use the correct way to calculate euler angles

If @kriswiner has something to add I will be glad to hear from him.
Nevertheless, I hope the guide helps! And if something is needed I will be glad to help :)

Best regards!

kriswiner, 2020-07-18:
> The code that you have posted before in ( #14 (comment)
>    1. start playing with beta value and understand how beta plays a role
>    2. confirm that the euler angles are as expected. Some versions of the

Only that if you run the gyro at 250 Hz your mcu needs to be able to
iterate 10 - 20 times per new gyro data sample to converge. So your mcu
must be able to achieve 2 -3 kHz fusion rate. No AVR mcu can do this.

What fusion rate do you obtain?

j-mcc1993, 2020-07-19:
Hi all,

I made the changes in the example to adapt @kriswiner's example to the Nano 33 boards. You can find the updated code [here](https://github.com/j-mcc1993/LSM9DS1/blob/master/LSM9DS1_BasicAHRS_Nano33.ino).

I hope this will be of some help!

foxy-munich, 2020-11-10:
I read almost all of the comments but i never get a working solution.  Is there anybody with an arduino nano 33 ble sense who get yaw pitch roll in degrees and can post his sketch?

Best regards

kriswiner, 2020-11-10:
> So in your project which way is your nano 33 ble facing? Mine was on its
> side turned to the left, so left was down and right was up etc. because of
> this I had to use different signs for my XYZ. If yours is the same way I
> can show you how I did mine, I'll post it later today, but if not I
> recommend experimenting with different combinations of positive and
> negative X Y and Z as well as using the madgwickAHRS library as I described
> in some of my earlier posts. Good Luck and lmk if you have any further
> questions!

@foxy-munich

Can't get quaternions? Can't get accurate heading? What is the problem
exactly?

The sketches available here and at Adafruit and at Sparkfun and elsewhere
are never going to be exactly what you need for your specific hardware
situation. You are going to have to make modifications to suit.

Is this just a "can't make the Arduino Nano work" kind of problem? Maybe an
Arduino Nano forum would be more appropriate?

[51 more replies]