kriswiner/LSM9DS1 #17: LSM9DS1 Heading and Calibration

Shmanman, 2021-03-15:
Hi Kris,

First, Thank you for your forums and support.
We are building a new product using the LSM9DS1 IMU unit.
We try to use it on a slow flight drone and currently we are at a preliminary stage of sensitivity and reliability tests. 
I'm guessing it's an easy solution for you :-), but currently all we are trying to achieve is stable direction when holding the drone at a static phases and turning it (by hand) every 30-40 degrees.
For the code we use the madgwick filter.
Every 30 seconds after start up we implement a 'hand movement' calibration (moving it at all 3 dimensions).
We also perform a gyro bypass correction for the first 100 measurements to correct the drift.
At this point we struggle with several issues:

- Updating time - when we move it to an heading, we see the heading updating very very slowly and it takes several seconds to reach to required one.
- Unstable heading: Sometimes it a constant drift and sometimes its +-40 degrees

Any help will be appreciated.
Thank you!
Asher.

kriswiner, 2021-03-15:
LSM6DS1 is way downrev, meaning LSM6DSM + MMC5983A offer far superior
sensor quality.

Then, the sensors need calibration, especially the mag.

These two are essential ingredients for accurate absolute orientation
estimation.

Lastly, the fusion filter requires a fast MCU, like an STM32L4 that we use
or a Teensy 3.2, for example. The idea is to run the sensors at a data rate
sufficient to capture the motion. In your case, 200 hz is sufficient for
gyro and accel. Then the fusion rate needs to be 10 - 20x this rate, so 2 -
3KHz. What MCU are you using?

Shmanman, 2021-03-18:
Hi Kris.
Thank you very much for your fast response.
For MCU we use STM32L4. 
Our sample rate is running at 200hz.
Currently, the LSM6DS1 is mounted to our prototype and we would like to try to improve it's measurements accuracy the best we can. (we are using SPARKFUN SEN-13944)
After additional sessions we've reached this point:
We can see the angular rate on the gyro measurement updating fast however the calculated angle (or heading) isn't updating accordingly. The madgwick filter is getting it's inputs from accellometer, gyro and magnetometer.
At this point (and due to the availability of the sparkfun card) we might change our IMU component.
Can you recommend on one for indoor slow flight drone?
Thank you!

kriswiner, 2021-03-18:
Well, let's see if  you are at the limit of what is possible with the
LSM6DS1.

You are using an L4, so what fusion rate are you running at?

How do you calibrate your sensors?

How do you check that they are really calibrated?

And how are you enforcing NED or ENU convention in the filter?

If these items are managed properly you should get reasonably fast response
(low latency) and ~4 degree heading accuracy.

Once you are there, then you can consider moving to much better sensors
like the LSM6DSM and MMC5983A, as mentioned.