kriswiner/LSM9DS1 #19: Adafruit NXP Library for LSM9DS1

meisamsafavi, 2021-10-11:
Hi to all.

First of all, I want to appreciate @kriswiner for making this code, that's a huge one, it makes using the sensor so easy and understandable for me. Thanks again.
Currently I'm trying to use the adafruit NXP Library for LSM9DS1, but unfortunately it's not working correctly.
I'm having some yaw drift problem with it.

I'm gonna be grateful for any help.

Regards,
Mason

kriswiner, 2021-10-11:
Need a bit more info to understand what problems you are having.

meisamsafavi, 2021-10-12:
As I said I'm using your LSM9DS1 Library for my Arduino Nano BLE Sense, and your IMU calibration method too. How should make sure the calibration is done correctly?
I have already tested the different sample rates, doesn't make any difference.

Regards

meisamsafavi, 2021-10-12:
I didn't think at all the solution might be easy at all.
After calibrating the sensors, mag and accel is OK but unfortunately gyro won't give zero values. 
I did calibration with your code and Femme Verbeek's code. non gives zeros.
any suggestions?

Regards

kriswiner, 2021-10-19:
> Hi again, @kriswiner <https://github.com/kriswiner>
> I'm steel working on this issue, after making sure the calibration has
> done properly, I realized it's all related to Kalman Covariance Variables
> and it should be personalized for any specific sensor.
> It was just an update, I'm gonna share the exact settings.
>
> Regards,
>
> Mason

Sorry I can't help you there...you could try the Madgwick fusion filter
just as a sanity check.