kriswiner/LSM9DS1 #20: LSM9DS1 Gyro lsb value

aktorh, 2022-01-09:
According to the datasheet, lsb of gyro is 8.75 mdps/lsb for +-245 dps.  245/32768 gives 7.4768 mdps. Its also different for +-500 and +-2000 dps. Which one we should use?

kriswiner, 2022-01-09:
Use the one stated in the data sheet. The reason for the difference is that
the scale factor is apparently non-linear. But I don;t really understand
why this should be the case. According to ST, one should use their spec
numbers...but you can ask them yourself for the why...

kriswiner, 2022-01-09:
Sorry, the reason is likely that 245 dps is not exactly right for the full
range also...so probably closer to 286 dps full range, etc...again, ask ST
what;s going on here...

aktorh, 2022-01-09:
OK. i will ask them :)

aktorh, 2022-01-11:
While waiting answer from ST , i checked official ST drivers and found this.
float_t lsm9ds1_from_fs245dps_to_mdps(int16_t lsb)
{
  return ((float_t)lsb * 8.75f);
}

float_t lsm9ds1_from_fs500dps_to_mdps(int16_t lsb)
{
  return ((float_t)lsb * 17.50f);
}

float_t lsm9ds1_from_fs2000dps_to_mdps(int16_t lsb)
{
  return ((float_t)lsb * 70.0f);
}

from source file (line 113)
https://github.com/STMicroelectronics/lsm9ds1/blob/2d4d0282f4628c4eee820d55a48b8d3d8494e7c3/lsm9ds1_reg.c

aktorh, 2022-01-12:
I got the answer from stm :
"Hi, you can refer to the datasheet or to the C drivers on Github.

In order to convert from LSB to physical units (in this case dps), you have to multiply the raw data (in LSB, obtained concatenating for example the OUT_X_H_G and the OUT_X_L_G, and casting the result into int16_t). Then, you have simply to multiply for 8.75, since the sensitivity is already expressed in milli-dps (so, no need to divide for 1000)

float_t lsm9ds1_from_fs245dps_to_mdps(int16_t lsb)
{
  return ((float_t)lsb * 8.75f);
}"

kriswiner, 2022-01-12:
Yes, same answer I got. But the unanswered question is why? ANyway, the
LSM6DS1 is way obsolete. The LSM6DSM is a much better accel/gyro.